Unsigned-Long/iKalibr #2: name suggestion

changh95, 2024-08-05:
(Not strictly related to the code)

If your calibration framework is inspired by Kalibr, and supports calibrating all kinds of sensor configurations (i.e. X-configuration)...

Why not call it xKalibr, like the sword [excalibur](https://en.wikipedia.org/wiki/Excalibur)? Just a suggestion! :smile:

Unsigned-Long, 2024-08-05:
Thanks for your issue :smiley:. In fact, the reason why `iKalibr` is called `iKalibr` has been explained clearly [here](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/why_ikalibr.md) (including the origin of the first letter `i` in `iKalibr` and the relationship between `iKalibr` and `Kalibr`). In addition, please note that the only similarity between `iKalibr` and `Kalibr` is that they use the continuous-time estimation method (the underlying estimation method), and almost everything else is different.

changh95, 2024-08-05:
Fully understood!
Thank you for your reply :)

Closing the issue as solved!

closed 2024-08-05 by changh95