Unsigned-Long/iKalibr #6: about the environment

zhuoyuan00, 2024-08-20:
Dear UL, thanks for your amazing work first. Can you show the detailed version of the dependencies including cmake,eigen,ceres,sophus and pangolin? I always met problems while installing ctraj and the error is /root/tools/algorithm_ws/src/ikalibr/thirdparty/ctraj/src/include/ctraj/factor/marginalization_factor.h:27:26: error: 'Manifold' in namespace 'ceres' does not name a type
   27 |             const ceres::Manifold *manifold;
Thanks very much.

zhuoyuan00, 2024-08-20:
this problem has been solved after testing several versions, thanks.

Unsigned-Long, 2024-08-21:
Thank you for your support :smiley:. A detailed installation guide can be found [here](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr.md). As mentioned in the tutorial, the `Ceres` **version needs to be 2.2.0** to support the `Manifold` module. Other dependent third-party libraries are the default versions of the system. My system is **Ubuntu 20.04**.

closed 2024-08-21 by Unsigned-Long