Unsigned-Long/iKalibr #8: EGL issue in docker

navegar69420, 2024-08-30:
**Describe the bug**
iKalibr seems unable to find the EGL instance when running ROS Noetic in a docker container. Host computer is Ubuntu 22.04 -- have tried first on a Jetson AGX Orin and also on a x86_64 laptop, both with the same error. 
I am wondering if anyone has ran into this before and has any suggestions on fixes. Below are two errors. The first doesn't stop the program, the second does. Strangely enough, the GUI window still opens and I can even see the lidar pointcloud a few minutes in. 

**To Reproduce**
Steps to reproduce the behavior:
1. On Ubuntu 22.04 laptop, get ros noetic docker image 
2. Install all necessary iKalibr dependencies from source as recommended in the Dockerfile and docker build
3. catkin_make, source /opt/ros/noetic.sh and devel/setup.bash, roscore
4. Set up the config file correctly
5. In particular, my docker run is as follows
`xhost +
docker run -it  \
  -v ~/radar_drone:/workspaces/isaac_ros-dev \
  -e DISPLAY=$DISPLAY \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  --network host \
  --privileged \
  --name container1 myapp:v1
`

**Expected behavior**
Expect iKalibr to run without issues. I'd gladly turn of the GUI, as all I need are the extrinsic calibrations and time offsets between my sensors.

**Screenshots**
[image]
[image]

**Desktop**
 - Windows laptop dual booted on Linux. 16 GB RAM, NVIDIA GeForce RTX 3060, Intel i7

Thank you for your help!

Unsigned-Long, 2024-08-30:
This answer is not a solution itself, but a guess at the solution. From your description and screenshots, it seems that the problem occurs in view rendering. `iKalibr`'s viewer is a secondary development based on `Pangolin`. In `Pangolin`'s official git [repository](https://github.com/stevenlovegrove/Pangolin.git), there are reports about problems when using `Pangolin` in docker. There are more detailed descriptions of the problems and solutions from the developers. I hope you can find some inspiration.
[image]

Unsigned-Long, 2024-08-30:
If the problem still exists, when we are free, we will add the option of `whether to enable GUI` to the configuration file of `iKalibr`, to flexibly adapt to different operating environments.

Unsigned-Long, 2024-12-20:
Here is a successfully tested example (using `Docker` to deploy `iKalibr`), and I hope you find it helpful.
https://github.com/Unsigned-Long/iKalibr/issues/14#issuecomment-2555846247

closed 2024-12-20 by Unsigned-Long