Unsigned-Long/iKalibr #9: Build error: 'fmt::format_string’ has not been declared

meitiever82, 2024-09-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. catkin build ikalibr

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Unsigned-Long, 2024-09-22:
Your problem may be caused by the fact that both `Sophus` and `Spdlog` libraries rely on `fmt`. Specifically, when compiling `Sophus`, specify to use the internal logging library instead of `fmt`. We have already detailed this problem clearly in the environment configuration document of `iKalibr`. If this is the problem, please refer to the [document](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr.md) (the `key points` part of `Install Required Third Libraries` section).

" both `spdlog` and `Sophus` would involve the `fmt` library, and would lead to conflict if the versions of involved `fmt` in `spdlog` and `Sophus` are different. To solve this, we recommend that you set the cmake option `SOPHUS_USE_BASIC_LOGGING` as `ON` when compile `Sophus`, this would avoid to involve `fmt` logger dependency in `Sophus`."

meitiever82, 2024-09-23:
Have you test build this tool on ubuntu 20.04? I set SOPHUS_USE_BASIC_LOGGING = ON when compile Sophus，and the error - ‘fmt::format_string’ has not been declared raised, which come from status.hpp
template <typename... T>
IKalibrStatus Status(Status flag, fmt::format_string<T...> fmt, T &&...args)
i have fixed this bug and the following error raised again:
/usr/bin/ld: CMakeFiles/ikalibr_raw_inertial_to_bag.dir/Unity/unity_0_cxx.cxx.o: in function `fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char>::on_error(char const*) [clone .isra.0]':
unity_0_cxx.cxx:(.text+0x1f): undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
Hope you can provide a better solution. thanks.

Unsigned-Long, 2024-09-23:
`iKalibr` is developed under `Ubuntu 20.04`. Except for the libraries (such as `Ceres`) that have been noted in the installation tutorial [document](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr.md), all other third-party libraries use the default library version. For example: in Ubuntu 20.04, the default installed `fmt` library version is `6.1.2`.

Therefore, arbitrarily modifying the library version may cause compilation errors (a problem may be solved after changing the version, but another problem will arise). You can start checking from this aspect.

Unsigned-Long, 2024-09-23:
Under `Ubuntu 20.04`, we have tested `catkin_make` and `catkin build`, and there are no problems.

meitiever82, 2024-09-24:
could we get in touch on WeChat or QQ？ and take a look remotely， thanks

Unsigned-Long, 2024-10-02:
The `fmt` conflict may be related to the `spdlog` library. I suggest you try to compile the `spdlog` library from [source code](https://github.com/gabime/spdlog.git) (just compile and install it directly), so that `spdlog` will use the internal `fmt` library.

Unsigned-Long, 2024-10-04:
This problem has been mentioned in #10 and solved, thus this issue will be closed.

closed 2024-10-04 by Unsigned-Long