Unsigned-Long/iKalibr #10: Build Issue due to ufomap

TitouanTyack, 2024-10-01:
Hi,
Thanks for sharing your work.

I have a build issue because of ufomap. I'm on Ubuntu 20.04 - ROS Noetic.

While running catkin_make -j4 -DUSE_CMAKE_UNITY_BUILD=ON, i have :
/home/titouan/iKalibr/src/ikalibr/include/viewer/viewer.h:45:10: fatal error: ufo/map/surfel_map.h: No such file or directory
   45 | #include "ufo/map/surfel_map.h"
I guess it is unable to find ufo dependancy. I checked, i think I'm on the good branch :
~/iKalibr/src/ikalibr/thirdparty/ufomap$ git branch gives: * (HEAD detached at b9de354)
ls src/ikalibr/thirdparty/ufomap-install/include/ufo/map/ returns:
code   empty        io.h      key.h        occupancy  octree_map_base.h  point.h    ray_caster  signed_distance  surfel_map.h  types.h
color  integration  iterator  node_base.h  octree     point_cloud.h      predicate  semantic    surfel           time          ufomap.h

For your information i didn't build colmap from source, 
- I used : sudo apt-get install colmap
- I built Sophus using SOPHUS_USE_BASIC_LOGGING=ON

If you have any idea of how to solve this issue, I would appreciate it,
Thanks

Unsigned-Long, 2024-10-01:
Yeah, our test shows that this problem does exist. The main reason may be related to the header file installation configuration in the cmakelists file of `ufomap`. After installation, library files of `ufomap` can be found through `find_package`, but the specific path of the header file cannot be found, so the header file path of `ufomap` needs to be given manually. Specifically, two lines need to be added after `find_package(ufomap)` in the cmakelists file of ikalibr (line 176):
```cmake
set(ufomap_INCLUDE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/thirdparty/ufomap-install/include)
include_directories(${ufomap_INCLUDE_DIR})
```
In fact, you can update `ikalibr` directly through `git pull`, since the bug has been fixed and uploaded to `github`, **you only need to pull the updated code and recompile.**

Unsigned-Long, 2024-10-02:
The `fmt` conflict may be related to the `spdlog` library. I suggest you try to compile the `spdlog` library from [source code](https://github.com/gabime/spdlog.git) (just compile and install it directly), so that `spdlog` will use the internal `fmt` library.

TitouanTyack, 2024-10-03:
I built successfully iKalibr !
After doing the procedure explained in https://github.com/Unsigned-Long/RIs-Calib/issues/1#issuecomment-2391354270, I compiled colmap (version 3.8) from source.

Thanks

Unsigned-Long, 2024-10-03:
Congratulations 🎊 ! I think it's time for us to rewrite the readme file about environment installation of `iKalibr`, hahahaha 😄 .

closed 2024-10-03 by Unsigned-Long