Unsigned-Long/iKalibr #13: Question about b-splines in the paper

QiuYue-bit, 2024-10-31:
您好，我最近刚开始接触B样条，在阅读您的论文的时候，有提到：
"Specifically, given a series of linear scale control points(also known as knots)"
随后，在Basalt-Calib相关的论文《Efficient Derivative Computation for Cumulative B-Splines on Lie Groups》中也有提到这件事情
 ”control points(also known as knots)“，然而在我查阅B样条相关的书籍时，发现knots和controls point好像是两个不同的概念？
knots的数量可以由controls point的数量和B样条的阶数决定，knots决定了 每个子基函数的有效区域，而control points是对应的系数。

这让我觉得有一些迷惑，为什么knots这个可以被当做control points来使用，因此想请教一下您，谢谢！

Unsigned-Long, 2024-10-31:
我的理解是有点类似于函数里面的自变量和因变量。比如给定一条以时间为自变量的位置B样条，控制点（control points）就是决定曲线走向的那些点（决定曲线函数值的点）。而节点（knot）就是和你所说的那样，是决定子基函数的有效区域的“标记”，是关于时间这个自变量的量。结点值可以简单理解为控制曲线各段的参数化范围。在参数空间内，当时间t从一个结点值变到下一个时，控制点影响的组合方式会发生变化，定义曲线的局部形状。比如下面这个例子，是一条4阶的均匀B样条，每一个时间点的函数值受四个控制点的控制影响，当时间t在t=1这个节点之前时，函数值由最开始的4个控制点决定，当时间t在t=1这个节点之后时，函数值由最后的4个控制点决定（控制点索引往后移一位）。
[image]
在图中：

- 红色圆点代表控制点（Control Points），并用虚线连接，形成控制多边形。曲线的形状由这些控制点决定，但曲线通常不穿过这些点。
- 蓝色曲线为通过控制点生成的B样条曲线。
- 绿色圆点表示关键的结点（Knots）位置，它们标记了参数化空间中不同的区间，并决定曲线的分段和形状。

Unsigned-Long, 2024-11-01:
是的，control points是我们需要估计的量，对于均匀B样条而言，knots是可以被计算出来的，根据输入的时间戳和均匀B样条的时间间隔可以确定计算函数值时相关联的几个控制点。论文里可能写得不太准确，到时候有时间可以改改。

QiuYue-bit, 2024-11-01:
理解您的意思了，感谢您的耐心解答🙏 ：）