Unsigned-Long/iKalibr #14: Working Dockerfile?

schablonmassig, 2024-11-07:
Hi! First off, truly impressive work. Thank you for sharing. 

Im trying to make a Dockerfile, included below, but erroring out on the last part. 

```
# Use Ubuntu 20.04 as base image
FROM osrf/ros:noetic-desktop-full

# Prevent interactive prompts during package installation
ENV DEBIAN_FRONTEND=noninteractive

# Install basic tools and dependencies
RUN apt-get update \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y \
    build-essential \
    wget \
    python3-pip \
    lsb-release \
    gnupg2 \
    git \
    ros-noetic-cv-bridge \
    ros-noetic-velodyne \
    libcereal-dev \
    libspdlog-dev \
    libyaml-cpp-dev \
    libgoogle-glog-dev \
    libgflags-dev \
    libatlas-base-dev \
    #libeigen3-dev \
    libsuitesparse-dev \
    libpcl-dev

# Install newer CMake version (3.27.0 for example)
RUN wget https://github.com/Kitware/CMake/releases/download/v3.27.0/cmake-3.27.0.tar.gz && \
    tar -zxvf cmake-3.27.0.tar.gz && \
    cd cmake-3.27.0 && \
    ./bootstrap && \
    make -j$(nproc) && \
    make install && \
    cd .. && \
    rm -rf cmake-3.27.0 cmake-3.27.0.tar.gz

# Verify CMake version
RUN cmake --version

RUN  wget http://ceres-solver.org/ceres-solver-2.2.0.tar.gz \
    && tar zxf ceres-solver-2.2.0.tar.gz \
    && mkdir ceres-bin \
    && mkdir solver && cd ceres-bin \
    && cmake ../ceres-solver-2.2.0 -DEXPORT_BUILD_DIR=ON -DCMAKE_INSTALL_PREFIX="../solver" \
    && make -j$((`nproc`-2))  \
    && make install 

# Install Eigen 3.4.0
RUN git clone https://gitlab.com/libeigen/eigen.git && \
    cd eigen && \
    git checkout c1d637433e3b3f9012b226c2c9125c494b470ae6 && \
    mkdir build-eigen && \
    cd build-eigen && \
    cmake .. && \
    make install && \
    cd ../.. && \
    rm -rf eigen

# Clone and build Sophus with basic logging
RUN git clone https://github.com/strasdat/Sophus.git && \
    cd Sophus && \
    mkdir build && cd build && \
    cmake .. -DSOPHUS_USE_BASIC_LOGGING=ON && \
    make -j$(nproc) && \
    make install

# Install magic-enum
RUN git clone https://github.com/Neargye/magic_enum.git && \
    cd magic_enum && \
    mkdir build && cd build && \
    cmake .. && \
    make -j$(nproc) && \
    make install

# Install Pangolin dependencies first
RUN apt-get update && apt-get install -y \
    libglew-dev \
    libboost-dev \
    libboost-thread-dev \
    libboost-filesystem-dev \
    libegl-dev \
    libwayland-dev \
    libxkbcommon-dev \
    wayland-protocols \
    libc++-dev \
    libepoxy-dev \
    && apt-get install -y ninja-build \
    && rm -rf /var/lib/apt/lists/*

# Clone and build Pangolin with its submodules
RUN git clone --recursive https://github.com/stevenlovegrove/Pangolin.git && \
    cd Pangolin && \
    # Install recommended prerequisites
    ./scripts/install_prerequisites.sh recommended || true && \
    # Configure and build
    cmake -B build && \
    cmake --build build && \
    cd build && \
    make install

# Setup ROS environment
RUN echo "source /opt/ros/noetic/setup.bash" >> ~/.bashrc

# Create workspace and clone iKalibr
WORKDIR /root/iKalibr
RUN mkdir -p src && \
    cd src && \
    git clone --recursive https://github.com/Unsigned-Long/iKalibr.git ikalibr

# Add these environment variables before building OpenGV
ENV CXXFLAGS="-std=c++14" \
CFLAGS="-std=c++14"

# Build third-party dependencies
WORKDIR /root/iKalibr/src/ikalibr
RUN chmod +x build_thirdparty.sh && \
    ./build_thirdparty.sh

# Generate ROS messages and build iKalibr
WORKDIR /root/iKalibr
RUN /bin/bash -c "source /opt/ros/noetic/setup.bash && \
    catkin_make ikalibr_generate_messages && \
    catkin_make -j$(nproc) -DUSE_CMAKE_UNITY_BUILD=ON"

# Set the default command
CMD ["/bin/bash"]
```

The above file yields

```
104.6 make: *** [Makefile:146: all] Error 2
104.6 CMake Error at cmake_install.cmake:46 (file):
104.6   file INSTALL cannot find
104.6   "/root/iKalibr/src/ikalibr/thirdparty/opengv-build/lib/libopengv.a": No
104.6   such file or directory.
104.6
104.6
------
Dockerfile:128
--------------------
 127 |     WORKDIR /root/iKalibr/src/ikalibr
 128 | >>> RUN chmod +x build_thirdparty.sh && \
 129 | >>>     ./build_thirdparty.sh
 130 |
```

Any advice on how to keep going? Do you know of any other successfull build?

Unsigned-Long, 2024-11-08:
This error comes from `opengv`, which is a third-party library of `iKalibr`. The error should mean that `opengv` did not compile successfully (the compilation failed but `cmake --install` was used, so this error was reported). You can check the compilation of `opengv`, which is organized into the `build_thirdparty.sh` script file of `iKalibr`.

schablonmassig, 2024-12-19:
Hi again! I got it working by adapting your recommended 20.04 pipeline. For reference for anyone else, heres the dockerfile which worked for me. I had to add some pangolin- and ceres-related stuff, but it works just fine now on windows 10. 

```
FROM osrf/ros:noetic-desktop-full

# Install system dependencies
RUN apt-get update && apt-get install -y \
    build-essential \
    libcereal-dev \
    libyaml-cpp-dev \
    ros-noetic-cv-bridge \
    ros-noetic-velodyne \
    wget \
    git \
    libgoogle-glog-dev \
    libgflags-dev \
    libatlas-base-dev \
    libeigen3-dev \
    libsuitesparse-dev \
    libglew-dev \
    libboost-all-dev \
    libgl1-mesa-dev \
    libegl1-mesa-dev \
    libwayland-dev \
    libxkbcommon-dev \
    wayland-protocols \
    && rm -rf /var/lib/apt/lists/*

# Set working directory
WORKDIR /tmp

# Install CMake 3.27
RUN wget https://github.com/Kitware/CMake/releases/download/v3.27.0/cmake-3.27.0.tar.gz && \
    tar -zxvf cmake-3.27.0.tar.gz && \
    cd cmake-3.27.0 && \
    ./bootstrap && \
    make -j$(nproc) && \
    make install && \
    cd .. && \
    rm -rf cmake-3.27.0 cmake-3.27.0.tar.gz

# Install dependencies (Ceres, Sophus, magic_enum, Pangolin, spdlog)
RUN git clone --branch 2.2.0 --single-branch --recurse-submodules https://github.com/ceres-solver/ceres-solver && \
    cd ceres-solver && \
    mkdir build && cd build && \
    cmake .. && \
    make -j$(nproc) && \
    make install && \
    cd ../.. && \
    rm -rf ceres-solver

RUN git clone --branch 1.22.10 --single-branch https://github.com/strasdat/Sophus.git && \
    cd Sophus && \
    mkdir build && cd build && \
    cmake .. -DSOPHUS_USE_BASIC_LOGGING=ON && \
    make -j$(nproc) && \
    make install && \
    cd ../.. && \
    rm -rf Sophus

RUN git clone --branch v0.9.6 --single-branch https://github.com/Neargye/magic_enum.git && \
    cd magic_enum && \
    mkdir build && cd build && \
    cmake .. && \
    make -j$(nproc) && \
    make install && \
    cd ../.. && \
    rm -rf magic_enum

RUN git clone --branch v0.8 --single-branch --recursive https://github.com/stevenlovegrove/Pangolin.git && \
    cd Pangolin && \
    mkdir build && cd build && \
    cmake .. && \
    make -j$(nproc) && \
    make install && \
    cd ../.. && \
    rm -rf Pangolin && \
    ldconfig

RUN git clone https://github.com/gabime/spdlog.git && \
    cd spdlog && \
    mkdir build && cd build && \
    cmake .. && \
    make -j$(nproc) && \
    make install && \
    cd ../.. && \
    rm -rf spdlog

# Set up iKalibr workspace
WORKDIR /root/iKalibr/src
RUN git clone --recursive https://github.com/Unsigned-Long/iKalibr.git ikalibr

# Build thirdparty dependencies
WORKDIR /root/iKalibr/src/ikalibr
RUN chmod +x build_thirdparty.sh && \
    ./build_thirdparty.sh

# Build iKalibr
WORKDIR /root/iKalibr
RUN /bin/bash -c '. /opt/ros/noetic/setup.bash && \
    catkin_make ikalibr_generate_messages && \
    catkin_make -j8 -DUSE_CMAKE_UNITY_BUILD=ON'

# Set up the entrypoint
COPY entrypoint.sh /
RUN chmod +x /entrypoint.sh
ENTRYPOINT ["/entrypoint.sh"]

# Set the default command
CMD ["/bin/bash"]

#CMD ["roslaunch", "ikalibr", "ikalibr-learn.launch"]

```

closed 2024-12-19 by schablonmassig