Unsigned-Long/iKalibr #15: Shared Post - Issues and Solutions in the Compilation Process

cchester25, 2024-12-02:
I would like to thank the author for their outstanding work. To assist with the faster deployment of ikalib, I am sharing the issues and solutions I encountered during my own deployment process, hoping to help others in need.
**Describe the bug**
To avoid conflicts with installed libraries, I installed dependencies like Ceres, Spdlog, etc., in a non-root directory during the deployment of ikalib. 

**System Information**
- Operating System: Ubuntu 20.04  
- Installed Dependencies in Root Directory: Eigen 3.3.7, Ceres 1.14.0, Spdlog 1.5.0, Pangolin 0.8

**To Reproduce** Steps to reproduce the behavior:
1. **Source Compilation**
   - Follow the author's installation process to download and compile required dependencies from source.
     a. The provided CMake version is 3.27.0, but the minimum required version for installing Glomap is 3.28.0.
     b. I added `-DCMAKE_INSTALL_PREFIX="path/to/your"` during the CMake process to install dependencies in a custom location.
     c. Ceres 2.2.0 encountered an issue with CUDA not finding `/usr/bin/nvcc`. Add `-DCMAKE_CUDA_COMPILER="/usr/local/cuda/bin/nvcc"` after CMake.
     d. To compile Glomap with Eigen 3.3.7 and Ceres 1.14.0, change `Ceres::ceres` in CMakeLists.txt to `ceres`. For Eigen 3.4.0 and Ceres 2.2.0, first install Eigen 3.4.0, then recompile Ceres with Eigen 3.4.0 dependencies, and modify the CMakeLists.txt paths accordingly:
        ```
        set(Ceres_DIR path/to/your/lib/cmake/Ceres)
        find_package(Ceres 2.2.0 REQUIRED)
        set(CERES_INCLUDE_DIRS path/to/your/include/ceres)
        include_directories(${CERES_INCLUDE_DIRS})
        ```
        Finally, add `-DCMAKE_CUDA_COMPILER="/usr/local/cuda/bin/nvcc"` and compile Glomap.

2. **Compiling ikalib's Third-Party Libraries**
   - Pay special attention to the `ctraj` library. Modify `ctraj/src/CMakeLists.txt` as follows:
     ```
     set(Sophus_DIR path/to/your/share/sophus/cmake)
     find_package(Sophus 1.22.0 REQUIRED)
     set(SOPHUS_INCLUDE_DIRS path/to/your/include/sophus)
     include_directories(${SOPHUS_INCLUDE_DIRS})

     set(Ceres_DIR path/to/your/lib/cmake/Ceres)
     find_package(Ceres 2.2.0 REQUIRED)
     set(CERES_INCLUDE_DIRS path/to/your/include/ceres)
     include_directories(${CERES_INCLUDE_DIRS})
     ```

3. **Compiling ikalibr_generate_messages**
   - If Ceres::ceres is not found, it indicates that Ceres 2.2.0 could not be located. Add `-DCeres_DIR="path/to/your/lib/cmake/Ceres"` after `catkin_make`.

4. **Run Catkin Make**
   - Command: `catkin_make -j8 -DUSE_CMAKE_UNITY_BUILD=ON`
     a. If `magic_enum` is not found, modify `iKalib/src/ikalib/CMakeLists.txt`:
        ```
        set(magic_enum_DIR path/to/your/lib/cmake/magic_enum)
        find_package(magic_enum)
        set(MAGIC_ENUM_INCLUDE_DIRS path/to/your/include)
        include_directories(${MAGIC_ENUM_INCLUDE_DIRS})
        ```
     b. If `fmt::format_string` is undeclared, modify `iKalib/src/ikalib/CMakeLists.txt` by adding before Build:
        ```
        set(spdlog_DIR path/to/your/lib/cmake/spdlog)
        find_package(spdlog)
        set(SPDLOG_INCLUDE_DIRS path/to/your/include/spdlog)
        include_directories(${SPDLOG_INCLUDE_DIRS})
        ```

**Additional Context**
If CMake is not upgraded to version 3.27.0 or later, issues such as missing `CUDA::cusolver` may arise during the compilation of `ctraj`.

Unsigned-Long, 2024-12-03:
Thank you so much for sharing your experience and the solutions you encountered during the `iKalibr` deployment process! Your feedback is incredibly valuable, and it will certainly help others who may face similar issues. I appreciate the time and effort you've taken to help improve the deployment process, and I'm glad to hear that you found the library useful. If you have any further suggestions or run into any other issues, please don't hesitate to share them with us!

cchester25, 2024-12-05:
> Thank you so much for sharing your experience and the solutions you encountered during the `iKalibr` deployment process! Your feedback is incredibly valuable, and it will certainly help others who [...]
Hello, I noticed that the Imu accelerometer values in the ikalibr dataset seem to have been normalized. How can I restore the accelerometer values to m/s²?

closed 2024-12-20 by Unsigned-Long