Unsigned-Long/iKalibr #17: degenerate aware calibration

QMonteCarlo, 2024-12-20:
Hi, thanks for your excellent work to integrate so many calibration features in this repo, and I found it very convenient to use.
However, it is necessary to point out,  a lot of extrinsic calibration is performed for vehicles, and the motion is generally degenerate, which may not meet the requirement of sufficient motion excitement.

And there are indeed some works on this problem, we are able to at best calibrate 5 out of 6 DoF extrinsic parameters. Maybe in the future you could consider the degeneracy motion case?

Unsigned-Long, 2024-12-20:
Indeed, the `iKalibr` is a motion-based spatiotemporal calibrator, which requires sufficient motion excitation to ensure the observability of the spatiotemporal parameters. Therefore, for handheld or small-sized multi-sensor setups, achieving motion excitation is relatively easy. However, for larger sensor suites, such as those mounted on ground vehicles, it may be challenging to obtain sufficient motion excitation. :thinking: In such cases, observability analysis and constraints are essential to ensure the accuracy of parameter estimation. This feature may be supported in a future version of `iKalibr`, although due to my current commitments, the timeline may be delayed. Anyway, contributions to incremental development of `iKalibr` are always welcome! :clap:

Mikhail2017, 2025-02-23:
>to ensure the observability of the spatiotemporal parameters

Btw do you measure observability directly in your solvers? Maybe do you have any ideas on the ways to estimate it?

Unsigned-Long, 2025-02-27:
In fact, you may refer to Section V of this paper [Observability-Aware Intrinsic and Extrinsic Calibration of LiDAR-IMU Systems](https://ieeexplore.ieee.org/abstract/document/9787062), which employs TSVD to perform observability-aware estimation, updating only the states lying on the observable directions.