Unsigned-Long/iKalibr #18: Surfels only occuring in restricted volume

schablonmassig, 2024-12-20:
**Describe the bug**
The surfels should possible cover all of the point cloud. However, the surfels appears restricted to the positive axes , regardless of the quality of the cloud. 

**To Reproduce**
Steps to reproduce the behavior:
1. Perform calibration. 

Ive got three calibration bags. The same error occurs in all of them. 

**Expected behavior**
The surfel search only occurs in a volume which corresponds to the axes of the system. See images. Intuitively, it should instead cover the complete volume, in order to increase robustness. 

[image]
[image]
[image]

Unsigned-Long, 2024-12-20:
This seems rather strange, and I have not encountered such a situation before. Based on the image you provided, it can be inferred that when the grid-based map was constructed using `ufomap`, the points located along the negative axis were removed, leading to the fact that during the construction of the point-to-plane data association, only the grid points in the positive axis direction were considered (as shown in the figure). The gray points in the image represent the constructed global point cloud map, that's correct. :thinking: I suspect that something unexpected may have occurred during the construction of the plane feature map? (`ufomap`?)

schablonmassig, 2024-12-20:
Thank you for your thoughts. I thought id try do some print debugging along the lines you suggested. The included debug steps are as far as my limited C++ understanding can take me im afraid. 

1. Check that its not just some error when viewing in gui. The finished surfel file is similar to what I see in gui. In the gravity_aligned_map.pcd, I see a clear difference in crispness between the side with surfels and the one without. Thus the error is actually affecting the calculation.  
3. I began by checking InsertCloudToSurfelMap, in pts_association.h. The incoming cloud, `pclCloud` , are not truncated to positive axes, ie the bounds are positive to negative in x, y, and z. So far so good. 
4. The bounds of the surfel map is also good, as indiciated by the included printout below. 
5. I check it as well in calib_solver_da_impl.cpp, at line 286, before we pass it to viewer. The map extents appears good there as well. 

```
Point Cloud Bounds:
X: [-65.4543, 8.35473]
Y: [-31.3052, 36.9023]
Z: [-2.17899, 19.0003]

Surfel Map Statistics:
Number of Surfels: 33685
Total Points in Surfels: 113087774
Surfel Bounds:
X: [-59.7064, 8.35473]
Y: [-28.9995, 36.9023]
Z: [-2.17899, 19.0003]
```

Including the edited function for creating the surfel. 

```
void InsertCloudToSurfelMap(ufo::map::SurfelMap &map, pcl::PointCloud<PointType> &pclCloud) {
        int cloudSize = pclCloud.size();

        float minX = std::numeric_limits<float>::max();
        float minY = std::numeric_limits<float>::max();
        float minZ = std::numeric_limits<float>::max();
        float maxX = -std::numeric_limits<float>::max();
        float maxY = -std::numeric_limits<float>::max();
        float maxZ = -std::numeric_limits<float>::max();

        for (const auto& point : pclCloud.points) {
            minX = std::min(minX, point.x);
            minY = std::min(minY, point.y);
            minZ = std::min(minZ, point.z);
            maxX = std::max(maxX, point.x);
            maxY = std::max(maxY, point.y);
            maxZ = std::max(maxZ, point.z);
        }

        std::cout << "Point Cloud Bounds:" << std::endl;
        std::cout << "X: [" << minX << ", " << maxX << "]" << std::endl;
        std::cout << "Y: [" << minY << ", " << maxY << "]" << std::endl;
        std::cout << "Z: [" << minZ << ", " << maxZ << "]" << std::endl;

        ufo::map::PointCloud ufoCloud;
        ufoCloud.resize(cloudSize);

#pragma omp parallel for num_threads(omp_get_max_threads()) default(none) \
    shared(ufoCloud, cloudSize, pclCloud)
        for (int i = 0; i < cloudSize; i++) {
            ufoCloud[i].x = pclCloud.points[i].x;
            ufoCloud[i].y = pclCloud.points[i].y;
            ufoCloud[i].z = pclCloud.points[i].z;
        }

        map.insertSurfelPoint(std::begin(ufoCloud), std::end(ufoCloud));
        
        CheckSurfelMapBounds();
    }

    void CheckSurfelMapBounds() const {
        // Get all surfels that exist in the map
        auto pred = ufo::map::predicate::HasSurfel();
        auto nodes = _smp.query(pred);

        float minX = std::numeric_limits<float>::max();
        float minY = std::numeric_limits<float>::max();
        float minZ = std::numeric_limits<float>::max();
        float maxX = -std::numeric_limits<float>::max();
        float maxY = -std::numeric_limits<float>::max();
        float maxZ = -std::numeric_limits<float>::max();

        size_t totalPoints = 0;
        size_t numSurfels = 0;

        for (const auto& node : nodes) {
            numSurfels++;  // Count nodes as we iterate
            const auto& surfel = _smp.getSurfel(node);
            const auto& mean = surfel.getMean();
            
            minX = std::min(minX, static_cast<float>(mean.x));
            minY = std::min(minY, static_cast<float>(mean.y));
            minZ = std::min(minZ, static_cast<float>(mean.z));
            maxX = std::max(maxX, static_cast<float>(mean.x));
            maxY = std::max(maxY, static_cast<float>(mean.y));
            maxZ = std::max(maxZ, static_cast<float>(mean.z));
            
            totalPoints += surfel.getNumPoints();
        }

        std::cout << "\nSurfel Map Statistics:" << std::endl;
        std::cout << "Number of Surfels: " << numSurfels << std::endl;
        std::cout << "Total Points in Surfels: " << totalPoints << std::endl;
        std::cout << "Surfel Bounds:" << std::endl;
        std::cout << "X: [" << minX << ", " << maxX << "]" << std::endl;
        std::cout << "Y: [" << minY << ", " << maxY << "]" << std::endl;
        std::cout << "Z: [" << minZ << ", " << maxZ << "]" << std::endl;
    }
};
```

Unsigned-Long, 2024-12-21:
This is quite strange. I have just retested the `LI-Calib` `Court-01.bag` dataset (see [dataset description](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/dataset.md) and [data](https://drive.google.com/drive/folders/1kYLVLMlwchBsjAoNqnrwq2N2Ow5na4VD)), and there were no issues, as expected. I apologize, but I really can't pinpoint where the problem lies. :monocle_face: In the `iKalibr` configuration file, there is a `Preference::Outputs` option, which includes an output option for `LiDARMaps`. After the `iKalibr` processing is complete, both the lidar and surfel maps are output. You could try this and check if there is any issue with the output surfel map.
[image]

[image]

schablonmassig, 2024-12-21:
I got it working through a workaround in pts_association.cpp and viewer.cpp, by adding a shift to all in- and outgoing coordinates and thus ensuring association is occuring in positive coordinates. I'd by  happy to share the code if anyone else finds themselves with this error. Thanks again for taking a look, unsigned-long.

closed 2024-12-21 by schablonmassig

Huashuijingying, 2025-03-31:
> I got it working through a workaround in pts_association.cpp and viewer.cpp, by adding a shift to all in- and outgoing coordinates and thus ensuring association is occuring in positive coordinates. [...]

Hi,i found the same error.Can you share your pts_association.cpp and viewer.cpp? I would appreciate it very much.