Unsigned-Long/iKalibr #19: 在刚开始时结果还是合理的，但在后续优化中出现发散

Anutlet, 2025-01-13:
非常感谢您做的工作，对我的帮助很大。目前我在尝试标定自有数据，主要有两个livox_mid360和imu分别为lidar自带的一个和自己装配的imu，走8字采集的自有数据。在跑数据刚刚开始时，两个imu和lidar相对位置与实际情况是较为一致的，但在后续的处理过程中发散，最后发生乱漂的情况；这是我的参数配置问题吗，以下是我的配置，希望能够提供下指导；Configor:
  DataStream:
    IMUTopics:
      - key: "/imu"
        value:
          Type: "SENSOR_IMU"
          AcceWeight: 17.68
          GyroWeight: 57.66
      - key: "/livox/imu_0"
        value:
          Type: "SENSOR_IMU_G"
          Intrinsics: "/home/code/calib/ikalibr_ws/src/iKalibr/config/imu-intri_livox.yaml"
          AcceWeight: 09.00
          GyroWeight: 29.00
    RadarTopics:
    LiDARTopics:
      - key: "/livox/lidar_0"
        value:
          Type: "LIVOX_CUSTOM"
          Weight: 50.0
      - key: "/livox/lidar_1"
        value:
          Type: "LIVOX_CUSTOM"
          Weight: 50.0

    CameraTopics:
    RGBDTopics:
    ReferIMU: "/imu"
    BagPath: "/home/data/data_calib/calib_0113/20250113_lidar.bag"
    BeginTime: 5
    Duration: 50
    OutputPath: "/home/data/data_calib_new/calib_0113"
  Prior:
    # unit: m/s^2
    GravityNorm: 9.8
    # priori about spatiotemporal parameters, given by corresponding config file path
    SpatTempPrioriPath: ""
    # if sensor are hardware-synchronized, you could choose to fix temporal parameters
    # by setting this field to 'false'
    OptTemporalParams: true
    # the range where the time offsets would be optimized.
    # make sure this range contains the ground truth of time offsets
    # If you're not sure, make this field large, but this could lead to longer optimization time
    TimeOffsetPadding: 0.10
    # readout time padding for RS camera, make sure this range contains the ground truth
    ReadoutTimePadding: 0.01
    # leaf size when down sample the map using 'pcl::VoxelGrid' filter
    # note that this field just for visualization, no connection with calibration
    # for outdoor, 0.05 is suggested, and for indoor: 0.1 is suggested
    MapDownSample: 0.05
    # the time distance of two neighbor control points, which determines the accuracy
    # of the representation of the B-splines. Smaller distance would lead to longer optimization time
    # common choices: from '0.02' to '0.10'
    KnotTimeDist:
      SO3Spline: 0.05
      ScaleSpline: 0.05
    # when lidar is involved in the calibration framework, the ndt odometer is employed to recover pose roughly
    NDTLiDAROdometer:
      # 0.5 for indoor case and 1.0 for outdoor case
      Resolution: 1.0
      KeyFrameDownSample: 0.1
    LiDARDataAssociate:
      # leaf size when down sample the map using 'pcl::VoxelGrid' filter
      # note that this field just for visualization, no connection with calibration
      # for outdoor, 0.1 is suggested, and for indoor: 0.05 is suggested
      MapDownSample: 0.1
      # associate point and surfel when distance is less than this value
      PointToSurfelMax: 0.1
      # chose plane as a surfel for data association when planarity is larger than this value
      # range: 0.0-1.0, 0.5-1.0 is suggested
      PlanarityMin: 1.0
  Preference:
    # whether using cuda to speed up when solving least-squares problems
    # if you do not install the cuda dependency, set it to 'false'
    UseCudaInSolving: true
    # currently available output content:
    # ParamInEachIter, BSplines, LiDARMaps, VisualMaps, RadarMaps, HessianMat,
    # VisualLiDARCovisibility, VisualKinematics, ColorizedLiDARMap,
    # AlignedInertialMes, VisualReprojErrors, RadarDopplerErrors, RGBDVelocityErrors, LiDARPointToSurfelErrors
    # NONE, ALL
    Outputs:
      - LiDARMaps
      - VisualMaps
      - RadarMaps
      - VisualLiDARCovisibility
    # supported data output format:
    # 1. JSON
    # 2. XML
    # 3. YAML
    # 4. BINARY (not recommended)
    # do not dwell on it, we have provided an additional ros program to perform data format transform
    OutputDataFormat: "YAML"
    # number of thread to use for solving, negative value means use all valid thread to perform solving
    ThreadsToUse: -1
    # scale of splines in viewer, you can also use 'a' and 'd' keys to
    # zoom out and in splines in run time
    SplineScaleInViewer: 3.0
    # scale of coordinates in viewer, you can also use 's' and 'w' keys
    # to zoom out and in coordinates in run time
    CoordSScaleInViewer: 0.3
[image]

跑完就变成了这样
[image]

Unsigned-Long, 2025-01-13:
我建议检查一下IMU数据，比如配置文件里面的IMU类型（以G为单位？m/sec？用plotjuggler查看比较方便，设定IMU类型后对应的IMU读取代码可以在[这里](https://github.com/Unsigned-Long/iKalibr/blob/master/src/sensor/imu_data_loader.cpp)找到），还有采集数据时是否运动激励足够（不要平缓的8字行，快一点）？数据一般采集30sec就够了，时间越长解算速度越慢。

Anutlet, 2025-01-14:
> （

非常感谢您的回复，我通过plotjuggler查看了imu数据如下图，上面两个是外置imu的，下面两个是livox内置imu的；外置imu不是G为单位，角速度是rad/s；而livox内置imu是以G为单位，角速度应该也为rad/s，所以外置imu类型配置SENSOR_IMU，livox内置imu配置SENSOR_IMU_G; 运动激励由于设备在x,y方向移动不便，可能激励不太够；
[image]

Anutlet, 2025-01-14:
> 我建议检查一下IMU数据，比如配置文件里面的IMU类型（以G为单位？m/sec？用plotjuggler查看比较方便，设定IMU类型后对应的IMU读取代码可以在[这里](https://github.com/Unsigned-Long/iKalibr/blob/master/src/sensor/imu_data_loader.cpp)找到），还有采集数据时是否运动激励足够（不要平缓的8字行，快一点）？数据一般采集30sec就够了，时间越长解算速度越慢。

非常感谢您的回复，我通过plotjuggler查看了imu数据如下图，上面两个是外置imu的，下面两个是livox内置imu的；外置imu不是G为单位，角速度是rad/s；而livox内置imu是以G为单位，角速度应该也为rad/s，所以外置imu类型配置SENSOR_IMU，livox内置imu配置SENSOR_IMU_G; 运动激励由于设备在x,z方向移动不便，可能激励不太够；
[image]

Unsigned-Long, 2025-01-14:
下面是我们某次使用iKalibr标定实验中两个IMU的数据序列（上面两个是第一个IMU的加速度和比力，下面两个是另一个IMU的），可以用来对照着定性地判定一下运动激励程度
[image]

Anutlet, 2025-01-14:
> 下面是我们某次使用iKalibr标定实验中两个IMU的数据序列（上面两个是第一个IMU的加速度和比力，下面两个是另一个IMU的），可以用来对照着定性地判定一下运动激励程度 [...]

非常感谢您提供的参考，想再问下配置时这两个imu的类型应该是什么呢？   
 # key: IMU topic, value: IMU type. Supported IMU types are:
    #   1.              SBG_IMU: https://github.com/SBG-Systems/sbg_ros_driver.git
    #   2.           SENSOR_IMU: gyro unit (rad/s), acce unit (m/s^2), https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html
    #   3.         SENSOR_IMU_G: gyro unit (rad/s), acce unit (G)
    #   4.     SENSOR_IMU_G_NEG: gyro unit (rad/s), acce unit (-G)
    #   5.       SENSOR_IMU_DEG: gyro unit (deg/s), acce unit (m/s^2), https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html
    #   6.     SENSOR_IMU_DEG_G: gyro unit (deg/s), acce unit (G)
    #   7. SENSOR_IMU_DEG_G_NEG: gyro unit (deg/s), acce unit (-G)
根据您提供的数据，我发现两个imu的角速度应该都是rad/s，而且重力加速度两者应该都是m/s^2，只是上面是正值,下面这个是负值?那配置为SENSOR_IMU应该是有问题的？而我的外置imu也是负值，但没有对应的类型，是否需要添加新的类型呢？

Anutlet, 2025-01-15:
非常感谢您提供的参考，下面是是我重新采集的数据；但是效果依然和之前类似，刚开始时点云还可以对齐，后续优化过程中就出现发散的情况；
[image]

Unsigned-Long, 2025-01-16:
我建议你先用iKalibr进行IMU-only multi-IMU 标定，也就是多IMU的时空标定，不引入lidar，看看标定效果怎么样，如果效果可以的话，那么可能问题和lidar数据有关（比如时间戳？）。

closed 2025-03-05 by Unsigned-Long