Unsigned-Long/iKalibr #20: Cannot build the package

cmburgul, 2025-01-20:
Great work!

If I build only ikalibr_generate_messages (as mentioned in the "Tested install on 20.04" in build iKalibr step ~catkin_make ikalibr_generate_messages~) I cannot launch ikalibr-learn node. 
If I build the whole package, I get this error
"
In file included from /home/cmb/singularity/workspaces_ros1/ikalibr_ws/src/ikalibr/include/sensor/camera.h:38,
                 from /home/cmb/singularity/workspaces_ros1/ikalibr_ws/src/ikalibr/src/sensor/camera.cpp:35,
                 from /home/cmb/singularity/workspaces_ros1/ikalibr_ws/build/ikalibr/CMakeFiles/ikalibr_sensor.dir/Unity/unity_0_cxx.cxx:4:
/home/cmb/singularity/workspaces_ros1/ikalibr_ws/src/ikalibr/include/util/utils.h:53:10: fatal error: veta/type_def.hpp: No such file or directory
   53 | #include "veta/type_def.hpp"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

"
 - Device: [Ubuntu 20.04, ros noetic]

Are you yet to open source the complete code?

Unsigned-Long, 2025-01-21:
Hello, `veta` is a third-party library required by `iKalibr`. The compilation of this dependency, along with other third-party libraries, can be accomplished by executing the `build_thirdparty.sh` script (this is required before compiling `iKalibr`). For further details, please refer to [this file](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr.md).

cmburgul, 2025-01-21:
Thanks, a tested pipeline for installing in 20.04 has issues. 
 https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr.md#a-tested-install-pipeline-on-ubuntu-2004

I had to build CTraj from scratch separately from https://github.com/Unsigned-Long/CTraj/

I was able to build successfully!

closed 2025-01-21 by cmburgul

huanhexiao-alt, 2026-01-15:
Hello，
1、 https://github.com/Unsigned-Long/CTraj/  is unavailable, is https://github.com/liuxinren456852/CTraj OK?
2、there is no veta on github，could you offer me one？