Unsigned-Long/iKalibr #22: build issue:  immintrin.h: No such file or directory

Huashuijingying, 2025-02-27:
I followed the **Install Pipeline On Ubuntu 20.04**,
while doing

> catkin_make -j8 -DUSE_CMAKE_UNITY_BUILD=ON

error occured:
```
[ 38%] Building CXX object ikalibr/CMakeFiles/ikalibr_core.dir/Unity/unity_1_cxx.cxx.o
[ 50%] Built target ikalibr_generate_messages_eus
[ 62%] Built target ikalibr_generate_messages_lisp
[ 74%] Built target ikalibr_generate_messages_nodejs
[ 75%] Building CXX object ikalibr/CMakeFiles/ikalibr_learn.dir/Unity/unity_0_cxx.cxx.o
In file included from /home/cat/code_lib/iKalibr/src/ikalibr/thirdparty/ufomap-install/include/ufo/map/code/code.h:46,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/thirdparty/ufomap-install/include/ufo/map/octree/octree_base.h:47,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/thirdparty/ufomap-install/include/ufo/map/octree_map_base.h:46,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/thirdparty/ufomap-install/include/ufo/map/surfel_map.h:46,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/include/core/pts_association.h:41,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/include/nofree/ufomap_learner.hpp:38,
                 from /home/cat/code_lib/iKalibr/src/ikalibr/exe/nofree/learn.cpp:39,
                 from /home/cat/code_lib/iKalibr/build/ikalibr/CMakeFiles/ikalibr_learn.dir/Unity/unity_0_cxx.cxx:4:
/home/cat/code_lib/iKalibr/src/ikalibr/thirdparty/ufomap-install/include/ufo/map/key.h:49:10: fatal error: immintrin.h: No such file or directory
   49 | #include <immintrin.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ikalibr/CMakeFiles/ikalibr_learn.dir/build.make:76: ikalibr/CMakeFiles/ikalibr_learn.dir/Unity/unity_0_cxx.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3594: ikalibr/CMakeFiles/ikalibr_learn.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
i found a similar issue also in **ufo** ,but it didn't help.https://github.com/Unsigned-Long/iKalibr/issues/10

Unsigned-Long, 2025-02-27:
Please verify whether the `ufomap` library has been successfully compiled. Note that before compilation, it is necessary to **switch** to the appropriate **branch** of the cloned `ufomap` repository, refer to `git checkout origin/devel_surfel` in [build_thirdparty.sh](https://github.com/Unsigned-Long/iKalibr/blob/master/build_thirdparty.sh).

Huashuijingying, 2025-02-27:
已收到您的来信！（自动回复）

closed 2025-03-01 by Huashuijingying