Unsigned-Long/iKalibr #23: 'YAML Parsing failed - provided NVP (ScaleSplineType) not found'

Huashuijingying, 2025-03-01:
非常感谢您的工作！
我在下载iKalibr-data数据集到data文件夹后，更改了config-real.yaml文件中的所有文件路径，启动一个了`roscore`，随后运行了指令：
`rosrun ikalibr ikalibr_prog _config_path:="/home/zxy/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-52-11/config-real.yaml"`
出现了报错：

> [C]-[7025]-[09:49:49.327] The configuration file '/home/zxy/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-52-11/config-real.yaml' is outdated or broken, and can not be loaded in iKalibr using cereal!!! To make it right, please refer to our latest configuration file template released at https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/config_template_note.md, and then fix your custom configuration file. Detailed cereal exception information: 
> 'YAML Parsing failed - provided NVP (ScaleSplineType) not found'

请问如何解决呢

[image]

Huashuijingying, 2025-03-02:
在"data/data/real-world/iKalibr-data-2024-06-25-14-52-11/config-real.yaml"添加了一些缺少的字段（`ScaleSplineType: LIN_POS_SPLINE`，`RGBDTopics：`，`MapDownSample: 0.05`）后，可以成功运行了。

Huashuijingying, 2025-03-02:
when running on `tum/rs-seq-1` dataset

> [I]-[11178]-[22:15:13.741] loading calibration data...
> [I]-[11178]-[22:15:13.752] source data duration: from '4294967296.00000' to '0.00000'.
> [C]-[11178]-[22:15:13.752] unknown error happened: 'Time is out of dual 32-bit range'

how to fix this?

Unsigned-Long, 2025-03-05:
For the TUM dataset, I created another data bag instead of directly using the official bag file. Regarding the timestamp error issue, it may be due to differences in the way timestamps are stored, such as not being embedded in the header field.

closed 2025-03-05 by Unsigned-Long

Unsigned-Long, 2025-03-05:
[image]

[image]

PanZhichen, 2025-05-29:
iKalibr-data-2024-06-25-14-52-11 运行之后，提示：`initialize rotation 'SO3_CmToBr' failed, this may be related to insufficiently excited motion or bad images`，请问你也是这样吗？ @Huashuijingying

v-mehta, 2025-07-15:
@Unsigned-Long I bumped into this error as well while running the raw bags from the TUM dataset. Would it be possible to share the processed bag you created? One example would be great to reproduce your run. 
```
[I]-[11178]-[22:15:13.741] loading calibration data...
[I]-[11178]-[22:15:13.752] source data duration: from '4294967296.00000' to '0.00000'.
[C]-[11178]-[22:15:13.752] unknown error happened: 'Time is out of dual 32-bit range'
```