Unsigned-Long/iKalibr #24: Segmentation fault when running on ikalibr-data:  [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

Huashuijingying, 2025-03-04:
Thank you for the great work!
running on :  ikalibr-data-2024-06-25-14-55-10
sfm method: colmap
```
[I]-[198704]-[20:49:35.764] saving visual reprojection errors finished!
[I]-[198704]-[20:49:35.764] saving radar radar doppler errors to dir: '/home/hhu/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-55-10/residuals/doppler_error'...
[I]-[198704]-[20:49:35.967] saving radar doppler errors finished!
[I]-[198704]-[20:49:35.967] saving visual kinematics to dir: '/home/hhu/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-55-10/kinematics'...
[I]-[198704]-[20:49:35.967] create visual images with gravity for camera '/cam_gs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[12.3 s] 
[I]-[198704]-[20:49:48.745] create visual images with gravity for camera '/cam_rs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[6.6 s] 
[I]-[198704]-[20:49:55.640] create visual linear velocity images for camera '/cam_gs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[12.4 s] 
[I]-[198704]-[20:50:08.566] create visual linear velocity images for camera '/cam_rs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[6.6 s] 
[I]-[198704]-[20:50:15.516] create visual angular velocity images for camera '/cam_gs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[12.3 s] 
[I]-[198704]-[20:50:28.319] create visual angular velocity images for camera '/cam_rs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[6.5 s] 
[I]-[198704]-[20:50:35.120] saving visual kinematics finished!
[I]-[198704]-[20:50:35.120] saving consistency images to dir: '/home/hhu/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-55-10/consistency'...
[I]-[198704]-[20:50:35.120] verify consistency between LiDAR and camera '/cam_gs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[265.2 s] 
[I]-[198704]-[20:55:01.288] verify consistency for camera '/cam_gs/image_compressed' finished
[I]-[198704]-[20:55:01.288] verify consistency between LiDAR and camera '/cam_rs/image_compressed'...
▏P ███████████████████████████████████████████████████▏[100.0%]-[524/524]-[181.5 s] 
[I]-[198704]-[20:58:03.468] verify consistency for camera '/cam_rs/image_compressed' finished
[I]-[198704]-[20:58:03.469] verify consistency finished!
[I]-[198704]-[20:58:03.469] saving visual colorized map to dir: '/home/hhu/iKalibr/src/ikalibr/data/real-world/iKalibr-data-2024-06-25-14-55-10/maps'...
[I]-[198704]-[20:58:03.469] create colorized map using camera '/cam_gs/image_compressed'...
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[I]-[198704]-[20:58:03.965] performing colorizing, this would cost some time...
Segmentation fault (core dumped)
```

Huashuijingying, 2025-03-04:
same fault on  iKalibr-data-2024-06-25-14-52-11

Unsigned-Long, 2025-03-05:
[image]

utkuoguzman, 2025-03-23:
> [image]

I didn't understand. Can you clarify please?

Huashuijingying, 2025-03-23:
已收到您的来信！（自动回复）