Unsigned-Long/iKalibr #25: Ubuntu 22.04 and/or ROS2 Humble Support

utkuoguzman, 2025-03-10:
**Is your feature request related to a problem? No.
A clear and concise description of what the problem is.
Ubuntu 22.04 and ROS2 Humble is a standard now. Do you plan to move your beautiful solution to there?

**Describe the solution you'd like**
That it works on Ubuntu 22.04 and ROS2 Humble

**Describe alternatives you've considered**
That it works on Ubuntu 20.04 and ROS2 Foxy 

**Additional context**
Add any other context or screenshots about the feature request here.

Unsigned-Long, 2025-03-11:
Indeed, `iKalibr` currently only supports `ROS1`, but its `ROS` dependency is limited to the data loader module. Porting to `ROS2` is technically feasible (interface adjustments required). `ROS2` support will be implemented when time resources permit.

Unsigned-Long, 2025-04-10:
We are excited to announce the release of the **official iKalibr Docker image**! To make it easier for users to use the `iKalibr` project, we have built and published a Docker image that simplifies the setup process. With this Docker image, you can now easily run `iKalibr` in an isolated environment without worrying about dependency issues or complex configurations. For more details, please see [here](https://github.com/Unsigned-Long/iKalibr/blob/master/docs/details/build_ikalibr_docker.md).

- Pre-configured environment with all necessary dependencies.
- Hassle-free setup, with just a few commands to get started.
- Flexibility to run on any platform supporting Docker.
- Access to the latest version of `iKalibr`, easily kept up-to-date via Docker pull.

closed 2025-04-10 by Unsigned-Long