Unsigned-Long/iKalibr #29: Unable to replicate the calibration results on Li-calib dataset

utkuoguzman, 2025-04-10:
**Describe the bug**
I have tried iKalibr on LI-calib dataset. The resultant point cloud was simply a mess for every one I tried.

**To Reproduce**
Steps to reproduce the behavior:
1. Write a launch file such that it calls the yaml of Court-1, Court-2, Garage-1 dataset etc.
2. Fix some missing parts of their yaml files, such as "RGBDTopics:" tag missing and add "MapDownSample: 0.5" under "Prior:" for the code to work.
3. Call the launch file.

**Expected behavior**
I expected coherent point clouds of an environment. Instead if I don't increase the "NDTLiDAROdometer:"s "Resolution:" parameter, I get:

> I]-[19000]-[14:23:14.941] IMU topic: '/imu1/data', data size: '009158', time span: from '+000.00117' to '+022.89399' (s)
> [I]-[19000]-[14:23:14.941] IMU topic: '/imu2/data', data size: '009158', time span: from '+000.00050' to '+022.89251' (s)
> [I]-[19000]-[14:23:14.941] IMU topic: '/imu3/data', data size: '009157', time span: from '+000.00136' to '+022.89226' (s)
> [I]-[19000]-[14:23:14.941] LiDAR topic: '/velodyne_packets', data size: '000226', time span: from '+000.10087' to '+022.79337' (s)
> [I]-[19000]-[14:23:14.941] raw start time: '+1581317762.41392' (s), raw end time: '+1581317785.30816' (s)
> [I]-[19000]-[14:23:14.941] aligned start time: '+000.00000' (s), aligned end time: '+022.89424' (s)
> [I]-[19000]-[14:23:14.941] calib start time: '+000.05000' (s), calib end time: '+022.84424' (s)
> 
> [I]-[19000]-[14:23:14.941] create spline bundle: start time: '0.05000', end time: '22.84424', so3 dt : '0.05000', scale dt: '0.05000'
> [I]-[19000]-[14:23:15.042] fitting rotation b-spline...
> WARNING: Logging before InitGoogleLogging() is written to STDERR
> E0410 14:23:15.054251 19000 solver.cc:725] Terminating: Can't use DENSE_SCHUR with dense_linear_algebra_library_type = CUDA because support not enabled when Ceres was built.
> [I]-[19000]-[14:23:15.054] here is the summary:
> Ceres Solver Report: Iterations: -2, Initial cost: -1.000000e+00, Final cost: -1.000000e+00, Termination: FAILURE
> 
> [MultiViewer] press [ctrl+'s'] to save the current scene, [ctrl+'c'] for camera view, and [ctrl+'v'] for total viewer.
> E0410 14:23:15.095875 19000 solver.cc:725] Terminating: Can't use DENSE_SCHUR with dense_linear_algebra_library_type = CUDA because support not enabled when Ceres was built.
> [I]-[19000]-[14:23:15.095] here is the summary:
> Ceres Solver Report: Iterations: -2, Initial cost: -1.000000e+00, Final cost: -1.000000e+00, Termination: FAILURE
> 
> [I]-[19000]-[14:23:15.104] LiDARs are integrated, initializing extrinsic rotations of LiDARs...
> [I]-[19000]-[14:23:15.104] performing ndt odometer for '/velodyne_packets' for extrinsic rotation initialization...
> ▏P ███████████████████████████████████████████████████████▏[100.0%]-[226/226]-[14.2 s] [E]-[19000]-[14:23:29.534] initialize rotation 'SO3_LkToBr' failed, this may be related to the 'NDTResolution' of lidar odometer.
> [ikalibr_prog-1] process has finished cleanly
> log file: /home/uoguzman/.ros/log/92c0b2b8-15e9-11f0-b0a5-3f14bb64aa10/ikalibr_prog-1*.log
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

And if I increase it up to 2.1, I get:

> +----------------------------------------------------------+
> |     ██▓ ██ ▄█▀▄▄▄       ██▓     ██▓ ▄▄▄▄    ██▀███       |
> |    ▓██▒ ██▄█▒▒████▄    ▓██▒    ▓██▒▓█████▄ ▓██ ▒ ██▒     |
> |    ▒██▒▓███▄░▒██  ▀█▄  ▒██░    ▒██▒▒██▒ ▄██▓██ ░▄█ ▒     |
> |    ░██░▓██ █▄░██▄▄▄▄██ ▒██░    ░██░▒██░█▀  ▒██▀▀█▄       |
> |    ░██░▒██▒ █▄▓█   ▓██▒░██████▒░██░░▓█  ▀█▓░██▓ ▒██▒     |
> |    ░▓  ▒ ▒▒ ▓▒▒▒   ▓▒█░░ ▒░▓  ░░▓  ░▒▓███▀▒░ ▒▓ ░▒▓░     |
> |     ▒ ░░ ░▒ ▒░ ▒   ▒▒ ░░ ░ ▒  ░ ▒ ░▒░▒   ░   ░▒ ░ ▒░     |
> |     ▒ ░░ ░░ ░  ░   ▒     ░ ░    ▒ ░ ░    ░   ░░   ░      |
> |     ░  ░  ░        ░  ░    ░  ░ ░   ░         ░          |
> |                                          ░               |
> +----------+-----------------------------------------------+
> |  iKalibr | https://github.com/Unsigned-Long/iKalibr.git  |
> +----------+----------------+--------+---------------------+
> |  Author  | Shuolong Chen  | E-Mail | shlchen@whu.edu.cn  |
> +----------+----------------+--------+---------------------+
> [I]-[19499]-[14:42:33.351] loading configure from yaml file '/home/uoguzman/iKalibr/src/ikalibr/data/li-calib/Garage-01/config-real.yaml'...
> [I]-[19499]-[14:42:33.353] main fields of configor:
>                                     IMUTopics: /imu1/data /imu2/data /imu3/data 
>                                   RadarTopics: 
>                                   LiDARTopics: /velodyne_packets 
>                                  CameraTopics: 
>                                    RGBDTopics: 
>                                   EventTopics: 
>                          DataStream::ReferIMU: /imu1/data
>                           DataStream::BagPath: /home/uoguzman/iKalibr/src/ikalibr/data/li-calib-data/Garage-01.bag
[... 534 lines cut ...]
> -- --------------------------------------------------------------------------
> 
> [I]-[19499]-[14:42:56.724] build final lidar map and point-to-surfel correspondences...
> [I]-[19499]-[14:42:56.724] undistort scans for lidar '/velodyne_packets'...
> ▏P ███████████████████████████████████████████████████████▏[100.0%]-[246/246]-[2.5 s] 
> [I]-[19499]-[14:42:59.223] marge scans from lidar '/velodyne_packets' to map...
> [I]-[19499]-[14:43:00.108] perform point to surfel association for lidar '/velodyne_packets'...
> ▏P ███████████████████████████████████████████████████████▏[100.0%]-[246/246]-[1.0 s] 
> [I]-[19499]-[14:43:01.095] total point to surfel count for LiDARs: 11267
> [I]-[19499]-[14:43:01.106] Solving is finished! Focus on the viewer and press [ctrl+'s'] to save the current scene!
> [I]-[19499]-[14:43:01.106] Focus on the viewer and press ['w', 's', 'a', 'd'] to zoom spline viewer!
> [I]-[19499]-[14:43:01.107] saving maps to dir: '/home/uoguzman/iKalibr/src/ikalibr/data/li-calib-out/Garage-01/maps'...
> [I]-[19499]-[14:43:01.108] save global lidar surfel map...
> [I]-[19499]-[14:43:01.109] save global lidar surfel map as '/home/uoguzman/iKalibr/src/ikalibr/data/li-calib-out/Garage-01/maps/lidar/surfel_map.pcd'
> [I]-[19499]-[14:43:01.349] save global lidar map...
> [I]-[19499]-[14:43:01.472] save global lidar map as '/home/uoguzman/iKalibr/src/ikalibr/data/li-calib-out/Garage-01/maps/lidar/gravity_aligned_map.pcd'
> [I]-[19499]-[14:43:01.478] saving aligned inertial measurements to dir: '/home/uoguzman/iKalibr/src/ikalibr/data/li-calib-out/Garage-01/residuals/inertial_error'...
> [I]-[19499]-[14:43:04.193] saving aligned inertial measurements finished!
> [I]-[19499]-[14:43:04.193] solving and outputting finished!!! Everything is fine!!!

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Additional context**
Add any other context about the problem here.

Unsigned-Long, 2025-04-10:
To be honest, I'm not sure what the issue is on your end. I just tested it here, and there were no problems at all (I use `Court-01.bag` in `LI-Calib` dataset ).
[image]

here is the configuration file I used for solving:
```
Configor:
  DataStream:
    # key: IMU topic, value: IMU type. Supported IMU types are:
    #   1. SENSOR_IMU: https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html
    #   2.    SBG_IMU: https://github.com/SBG-Systems/sbg_ros_driver.git
    #   3. SENSOR_IMU_G: https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html (acce unit: G)
    #   4. SENSOR_IMU_G_NEG: https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html (acce unit: -G)
    IMUTopics:
      - key: "/imu1/data"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/imu1-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
      - key: "/imu2/data"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/imu2-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
      - key: "/imu3/data"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/imu3-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
    # key: radar topic, value: radar type. Supported radar types are:
    #   1.     AINSTEIN_RADAR: https://github.com/AinsteinAI/ainstein_radar.git
    #   2. AWR1843BOOST_RAW: https://github.com/Unsigned-Long/ti_mmwave_rospkg.git
    #   3. AWR1843BOOST_CUSTOM: https://github.com/Unsigned-Long/ti_mmwave_rospkg.git
    #   4.  POINTCLOUD2_POSV: 'sensor_msgs/PointCloud2' with point format: [x, y, z, velocity]
    #   5.  POINTCLOUD2_POSIV: 'sensor_msgs/PointCloud2' with point format: [x, y, z, intensity, velocity]
    #   6.  POINTCLOUD2_XRIO: 'sensor_msgs/PointCloud2' with x-RIO point format (see https://github.com/christopherdoer/rio.git)
    RadarTopics:
    #   1.        Velodyne LiDARs: VLP_16_PACKET, VLP_POINTS
    #   2.          Ouster LiDARs: OUSTER_POINTS
    #   3. Hesai Pandar XT LiDARs: PANDAR_XT_POINTS
    #   4.           Livox LiDARs: LIVOX_CUSTOM (the official 'xfer_format'=1, mid-360 and avia is recommend)
    LiDARTopics:
      - key: "/velodyne_packets"
        value:
          Type: "VLP_16_PACKET"
          Weight: 100.0
    RGBDTopics:
    # key: camera topic, value: camera type. Supported camera types:
    #   1. SENSOR_IMAGE_GS: https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Image.html\n"
    #   2. SENSOR_IMAGE_RS_FIRST: first-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Image.html\n"
    #   3. SENSOR_IMAGE_RS_MID: middle-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Image.html\n"
    #   4. SENSOR_IMAGE_RS_LAST: last-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Image.html\n"
    #   5. SENSOR_IMAGE_COMP_GS: https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CompressedImage.html\n"
    #   6. SENSOR_IMAGE_COMP_RS_FIRST: first-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CompressedImage.html\n"
    #   7. SENSOR_IMAGE_COMP_RS_MID: middle-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CompressedImage.html\n"
    #   8. SENSOR_IMAGE_COMP_RS_LAST: last-row exposure, https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CompressedImage.html\n"
    CameraTopics:
    ReferIMU: "/imu1/data"
    BagPath: "/home/csl/dataset/li-calib/Court-01.bag"
    # the time piece: [BegTime, BegTime + Duration], unit: second(s)
    # if you want to use all time data for calibration, please set them to negative numbers
    # Note that the 'BegTime' here is measured from the start time of bag
    # and is not related to the timestamp of the data recorded in the package.
    # [5, 28]
    BeginTime: 5
    Duration: 23
    OutputPath: "/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01"
  Prior:
[... 43 lines cut ...]
    # AlignedInertialMes, VisualReprojErrors, RadarDopplerErrors
    # NONE, ALL
    Outputs:
      - LiDARMaps
      - VisualMaps
      - RadarMaps
      - VisualLiDARCovisibility
      - VisualKinematics
      - ColorizedLiDARMap
      - AlignedInertialMes
      - VisualReprojErrors
      - RadarDopplerErrors
    # supported data output format:
    # 1. JSON
    # 2. XML
    # 3. YAML
    # 4. BINARY (not recommended)
    OutputDataFormat: YAML
    # number of thread to use for solving, negative value means use all valid thread to perform solving
    ThreadsToUse: -1
    # scale of splines in viewer, you can also use 'a' and 'd' keys to zoom out and in splines in run time
    SplineScaleInViewer: 3.0
    # scale of coordinates in viewer, you can also use 's' and 'w' keys to zoom out and in coordinates in run time
    CoordSScaleInViewer: 0.3
```

closed 2025-04-10 by Unsigned-Long

Unsigned-Long, 2025-04-12:
this is the output:
```
csl@ulong2:~/ros_ws/iKalibr/src/ikalibr$ roslaunch ikalibr ikalibr-prog.launch config_path:=/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/config-real.yaml
... logging to /home/csl/.ros/log/5e6f683a-1777-11f0-a7e0-c330be10ccea/roslaunch-ulong2-11971.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ulong2:33737/

SUMMARY
========

PARAMETERS
 * /ikalibr_prog/config_path: /home/csl/ros_ws/...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    ikalibr_prog (ikalibr/ikalibr_prog)

auto-starting new master
process[master]: started with pid [12001]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5e6f683a-1777-11f0-a7e0-c330be10ccea
process[rosout-1]: started with pid [12030]
started core service [/rosout]
process[ikalibr_prog-2]: started with pid [12037]
+----------------------------------------------------------+
|     ██▓ ██ ▄█▀▄▄▄       ██▓     ██▓ ▄▄▄▄    ██▀███       |
|    ▓██▒ ██▄█▒▒████▄    ▓██▒    ▓██▒▓█████▄ ▓██ ▒ ██▒     |
|    ▒██▒▓███▄░▒██  ▀█▄  ▒██░    ▒██▒▒██▒ ▄██▓██ ░▄█ ▒     |
|    ░██░▓██ █▄░██▄▄▄▄██ ▒██░    ░██░▒██░█▀  ▒██▀▀█▄       |
|    ░██░▒██▒ █▄▓█   ▓██▒░██████▒░██░░▓█  ▀█▓░██▓ ▒██▒     |
|    ░▓  ▒ ▒▒ ▓▒▒▒   ▓▒█░░ ▒░▓  ░░▓  ░▒▓███▀▒░ ▒▓ ░▒▓░     |
|     ▒ ░░ ░▒ ▒░ ▒   ▒▒ ░░ ░ ▒  ░ ▒ ░▒░▒   ░   ░▒ ░ ▒░     |
|     ▒ ░░ ░░ ░  ░   ▒     ░ ░    ▒ ░ ░    ░   ░░   ░      |
|     ░  ░  ░        ░  ░    ░  ░ ░   ░         ░          |
|                                          ░               |
+----------+-----------------------------------------------+
|  iKalibr | https://github.com/Unsigned-Long/iKalibr.git  |
+----------+----------------+--------+---------------------+
|  Author  | Shuolong Chen  | E-Mail | shlchen@whu.edu.cn  |
+----------+----------------+--------+---------------------+
[I]-[12037]-[16:23:12.479] loading configure from yaml file '/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/config-real.yaml'...
[I]-[12037]-[16:23:12.485] main fields of configor:
                                    IMUTopics: /imu1/data /imu2/data /imu3/data 
                                  RadarTopics: 
                                  LiDARTopics: /velodyne_packets 
                                 CameraTopics: 
                                   RGBDTopics: 
                                  EventTopics: 
                         DataStream::ReferIMU: /imu1/data
                          DataStream::BagPath: /home/csl/dataset/li-calib/Court-01.bag
                        DataStream::BeginTime: 5
                         DataStream::Duration: 23
                       DataStream::OutputPath: /home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01
                           Prior::GravityNorm: 9.8
                     Prior::OptTemporalParams: true
                     Prior::TimeOffsetPadding: 0.05
                    Prior::ReadoutTimePadding: 0.04
                         Prior::MapDownSample: 0.1
               Prior::KnotTimeDist::SO3Spline: 0.05
             Prior::KnotTimeDist::ScaleSpline: 0.05
          Prior::NDTLiDAROdometer::Resolution: 0.5
  Prior::NDTLiDAROdometer::KeyFrameDownSample: 0.1
  Prior::LiDARDataAssociate::PointToSurfelMax: 0.1
      Prior::LiDARDataAssociate::PlanarityMin: 0.6
[... 556 lines cut ...]
-- --------------------------------------------------------------------------
-- OTHER FIELDS
-- 
-- GRAVITY IN MAP: 'Gx': +000.000000, 'Gy': -000.000000, 'Gz': -009.800000
-- --------------------------------------------------------------------------

[I]-[12037]-[16:24:29.092] build final lidar map and point-to-surfel correspondences...
[I]-[12037]-[16:24:29.092] undistort scans for lidar '/velodyne_packets'...
▏P ███████████████████████████████████████████████████████████████████████████████████████████████████████████████▏[100.0%]-[226/226]-[2.5 s] 
[I]-[12037]-[16:24:31.628] marge scans from lidar '/velodyne_packets' to map...
[I]-[12037]-[16:24:33.599] perform point to surfel association for lidar '/velodyne_packets'...
▏P ███████████████████████████████████████████████████████████████████████████████████████████████████████████████▏[100.0%]-[226/226]-[3.9 s] 
[I]-[12037]-[16:24:37.532] total point to surfel count for LiDARs: 2872075
[I]-[12037]-[16:24:37.912] Solving is finished! Focus on the viewer and press [ctrl+'s'] to save the current scene!
[I]-[12037]-[16:24:37.912] Focus on the viewer and press ['w', 's', 'a', 'd'] to zoom spline viewer!
[I]-[12037]-[16:24:37.948] saving maps to dir: '/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/maps'...
[I]-[12037]-[16:24:38.159] save global lidar surfel map...
[I]-[12037]-[16:24:38.211] save global lidar surfel map as '/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/maps/lidar/surfel_map.pcd'
[I]-[12037]-[16:24:38.365] save global lidar map...
[I]-[12037]-[16:24:38.427] save global lidar map as '/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/maps/lidar/gravity_aligned_map.pcd'
[I]-[12037]-[16:24:38.463] saving aligned inertial measurements to dir: '/home/csl/ros_ws/iKalibr/src/ikalibr/data/li-calib/Court-01/residuals/inertial_error'...
[I]-[12037]-[16:24:41.332] saving aligned inertial measurements finished!
[I]-[12037]-[16:24:41.332] solving and outputting finished!!! Everything is fine!!!

```

PanZhichen, 2025-05-29:
@utkuoguzman Hi, have you figure out the root cause? I used the same 'config-real.yaml' @Unsigned-Long posted above(just change absolute path), and **run iKalibr in the official docker container**, but the result was very wrong:
```
[I]-[9530]-[11:16:16.987] loading configure from yaml file '/mnt/ikalibr_ws/src/iKalibr/data/li-calib/Court-01/config-real.yaml'...
[I]-[9530]-[11:16:16.990] main fields of configor:
                                    IMUTopics: /imu1/data /imu2/data /imu3/data 
                                  RadarTopics: 
                                  LiDARTopics: /velodyne_packets 
                                 CameraTopics: 
                                   RGBDTopics: 
                                  EventTopics: 
                         DataStream::ReferIMU: /imu1/data
                          DataStream::BagPath: /mnt/lio_ws/dataset/ikalibr/li_calib/Court-01.bag
                        DataStream::BeginTime: 5
                         DataStream::Duration: 23
                       DataStream::OutputPath: /mnt/ikalibr_ws/src/iKalibr/data/li-calib/Court-01
                           Prior::GravityNorm: 9.8
                     Prior::OptTemporalParams: true
                     Prior::TimeOffsetPadding: 0.05
                    Prior::ReadoutTimePadding: 0.04
                         Prior::MapDownSample: 0.1
               Prior::KnotTimeDist::SO3Spline: 0.05
             Prior::KnotTimeDist::ScaleSpline: 0.05
          Prior::NDTLiDAROdometer::Resolution: 0.5
  Prior::NDTLiDAROdometer::KeyFrameDownSample: 0.1
  Prior::LiDARDataAssociate::PointToSurfelMax: 0.1
      Prior::LiDARDataAssociate::PlanarityMin: 0.6
             Prior::LossForRadarDopplerFactor: 0.1
            Prior::LossForPointToSurfelFactor: 0.02
                   Prior::LossForReprojFactor: 1
              Prior::LossForOpticalFlowFactor: 30
                 Preference::UseCudaInSolving: true
                 Preference::OutputDataFormat: YAML
                          Preference::Outputs: NONE|LiDARMaps|VisualMaps|RadarMaps|VisualLiDARCovisibility|VisualKinematics|ColorizedLiDARMap|AlignedInertialMes|VisualReprojErrors|RadarDopplerErrors
[I]-[9530]-[11:16:17.990] initialize calibration parameter manager using configor...
[I]-[9530]-[11:16:17.993] initialize calibration parameter manager using configor finished.
[I]-[9530]-[11:16:17.994] the detail calibration parameters are below: 
-- -------------------------
-- calibration parameters --
-- --------------------------------------------------------------------------
-- EXTRI
-- 
-- IMU: '/imu1/data'
-- EULER_BiToBr: 'Xr': +000.000000, 'Yp': -000.000000, 'Zy': +000.000000
--   POS_BiInBr: 'Px': +000.000000, 'Py': +000.000000, 'Pz': +000.000000
-- 
-- IMU: '/imu2/data'
-- EULER_BiToBr: 'Xr': +000.000000, 'Yp': -000.000000, 'Zy': +000.000000
--   POS_BiInBr: 'Px': +000.000000, 'Py': +000.000000, 'Pz': +000.000000
-- 
-- IMU: '/imu3/data'
-- EULER_BiToBr: 'Xr': +000.000000, 'Yp': -000.000000, 'Zy': +000.000000
--   POS_BiInBr: 'Px': +000.000000, 'Py': +000.000000, 'Pz': +000.000000
-- 
-- LiDAR: '/velodyne_packets'
-- EULER_LkToBr: 'Xr': +000.000000, 'Yp': -000.000000, 'Zy': +000.000000
--   POS_LkInBr: 'Px': +000.000000, 'Py': +000.000000, 'Pz': +000.000000
-- 
-- --------------------------------------------------------------------------
-- TEMPORAL
-- 
-- IMU: '/imu1/data'
-- TO_BiToBr: +000.000000 (s)
-- 
-- IMU: '/imu2/data'
-- TO_BiToBr: +000.000000 (s)
-- 
-- IMU: '/imu3/data'
-- TO_BiToBr: +000.000000 (s)
-- 
-- LiDAR: '/velodyne_packets'
[... 68 lines cut ...]
[I]-[9530]-[11:16:18.177] calibration data info:
[I]-[9530]-[11:16:18.177] IMU topic: '/imu1/data', data size: '009158', time span: from '+000.00117' to '+022.89399' (s)
[I]-[9530]-[11:16:18.177] IMU topic: '/imu2/data', data size: '009158', time span: from '+000.00050' to '+022.89251' (s)
[I]-[9530]-[11:16:18.177] IMU topic: '/imu3/data', data size: '009157', time span: from '+000.00136' to '+022.89226' (s)
[I]-[9530]-[11:16:18.177] LiDAR topic: '/velodyne_packets', data size: '000226', time span: from '+000.10087' to '+022.79337' (s)
[I]-[9530]-[11:16:18.177] raw start time: '+1581317762.41392' (s), raw end time: '+1581317785.30816' (s)
[I]-[9530]-[11:16:18.177] aligned start time: '+000.00000' (s), aligned end time: '+022.89424' (s)
[I]-[9530]-[11:16:18.177] calib start time: '+000.05000' (s), calib end time: '+022.84424' (s)

[I]-[9530]-[11:16:18.177] create spline bundle: start time: '0.05000', end time: '22.84424', so3 dt : '0.05000', scale dt: '0.05000'
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.[MultiViewer] press [ctrl+'s'] to save the current scene, [ctrl+'c'] for camera view, and [ctrl+'v'] for total viewer.[I]
-[9530]-[11:16:18.219] fitting rotation b-spline...
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0529 11:16:18.227636  9530 solver.cc:725] Terminating: Can't use DENSE_SCHUR with dense_linear_algebra_library_type = CUDA because support not enabled when Ceres was built.
[I]-[9530]-[11:16:18.227] here is the summary:
Ceres Solver Report: Iterations: -2, Initial cost: -1.000000e+00, Final cost: -1.000000e+00, Termination: FAILURE

E0529 11:16:18.259132  9530 solver.cc:725] Terminating: Can't use DENSE_SCHUR with dense_linear_algebra_library_type = CUDA because support not enabled when Ceres was built.
[I]-[9530]-[11:16:18.259] here is the summary:
[... cut ...] 00e+00, Termination: FAILURE

[I]-[9530]-[11:16:18.267] LiDARs are integrated, initializing extrinsic rotations of LiDARs...
[I]-[9530]-[11:16:18.267] performing ndt odometer for '/velodyne_packets' for extrinsic rotation initialization...
▏P ████████████████████████████████████████████████████████████████████████████████████████████████▏[100.0%]-[226/226]-[24.3 s] [E]-[9530]-[11:16:43.019] initialize rotation 'SO3_LkToBr' failed, this may be related to the 'NDTResolution' of lidar odometer.
```

PanZhichen, 2025-05-29:
@utkuoguzman
**Big discovery!**  Everything just magically works when I set `UseCudaInSolving` to `false` in the `config-real.yaml`! And this setting is mentioned by @Unsigned-Long in #30.

Unsigned-Long, 2025-05-29:
The issue may have arisen due to insufficient interactivity in the program I wrote, haha. The image we provided does not support `CUDA`; therefore, when running `iKalibr` in a `Docker` environment, the `UseCudaInSolving` parameter should be set to `false`. If the execution is on the host machine and the host supports `CUDA`, setting `UseCudaInSolving` to `true` can accelerate the computation. I believe it would be more user-friendly if `iKalibr` could automatically detect the `CUDA` availability in the current environment and provide appropriate prompts or suggestions accordingly.