Unsigned-Long/iKalibr #32: Radar-IMU Calibration Yields Poor Results with Custom Dataset

clepz, 2025-06-02:
Hi,

I've been experimenting with the radar-IMU calibration feature in `ikalibr` using data collected from my own handheld platform similar to the platform that used for River Dataset. Unfortunately, the results are far from what I expected, with **translation errors reaching up to 20cm.**

Interestingly, when I run the calibration with the **River dataset**, I encounter no issues, and the results are very close to the evaluation values reported in the paper.

Here are the methods I used for data collection:
1. **Indoor Parking Lot - Figure-Eight (Fast):** Rapidly moving the platform in a figure-eight pattern within an indoor parking garage.
1. **Indoor Parking Lot - Figure-Eight & Circles:** Moving the platform in a figure-eight pattern and then completing circles at the current location within an indoor parking garage.
1. **Office Environment - Figure-Eight (Fast):** Rapidly moving the platform in a figure-eight pattern at my current location within an office environment.

Although the error rates are generally very high, the **closest (albeit still inaccurate) results were obtained with the data from the third method** (office environment, fast figure-eights).

Could you provide any insights into what might be causing this discrepancy or suggest potential areas to investigate?

Thanks.

### Sensors:
- **Radar:** 
  - ros repository from [here](https://github.com/radar-lab/ti_mmwave_rospkg)
  - AWR1843Boost
  - out of the box demo binary
- **IMU**:
  - Xsens MTI-300 ahrs
  - 400Hz
  - ros driver coming from mt software suit
  - configuration: [image]

### Sensor placements

[image]

here is the calibration configuration:
```yaml
Configor:
  DataStream:
    IMUTopics:
      - key: "/imu/data"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/home/iKalibr/src/ikalibr/config/imu-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
    RadarTopics:
      - key: "/ti_mmwave/radar_scan"
        value:
          Type: "AWR1843BOOST_RAW"
          Weight: 10.0
      - key: "/ti_mmwave/radar_scan1"
        value:
          Type: "AWR1843BOOST_RAW"
          Weight: 10.0
      - key: "/ti_mmwave/radar_scan2"
        value:
          Type: "AWR1843BOOST_RAW"
          Weight: 10.0
    LiDARTopics:
    RGBDTopics:
    CameraTopics:
    ReferIMU: "/imu/data"
    BagPath: "/record1.bag"
    BeginTime: -1
    Duration: -1
    OutputPath: "/calibration_output/test1"
  Prior:
    GravityNorm: 9.8
    SpatTempPrioriPath: ""
    OptTemporalParams: true
    TimeOffsetPadding: 0.50
    ReadoutTimePadding: 0.01
    KnotTimeDist:
      SO3Spline: 0.05
      ScaleSpline: 0.05
    NDTLiDAROdometer:
      Resolution: 0.5
      KeyFrameDownSample: 0.1
    MapDownSample: 0.05
    LiDARDataAssociate:
      MapDownSample: 0.05
      PointToSurfelMax: 0.1
      PlanarityMin: 0.6
    CauchyLossForRadarFactor: 0.1
    CauchyLossForLiDARFactor: 0.5
    CauchyLossForCameraFactor: 1.0
  Preference:
    UseCudaInSolving: false
    Outputs:
      - RadarMaps
      - VisualKinematics
      - AlignedInertialMes
      - VisualReprojErrors
      - RadarDopplerErrors
    OutputDataFormat: YAML
    ThreadsToUse: 16
    SplineScaleInViewer: 3.0
    CoordSScaleInViewer: 0.3
```

here is the rosbag info output that collected via 3rd method mentioned above:
```bash
duration:    1:51s (111s)
size:        23.2 MB
messages:    110059
compression: none [30/30 chunks]
types:       ikalibr/AWR1843RadarScan [7a726cbc7d2934bb55d96dada9040f86]
             sensor_msgs/Imu          [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /imu/data                44537 msgs    : sensor_msgs/Imu         
             /ti_mmwave/radar_scan    17105 msgs    : ikalibr/AWR1843RadarScan
             /ti_mmwave/radar_scan1   23798 msgs    : ikalibr/AWR1843RadarScan
             /ti_mmwave/radar_scan2   24619 msgs    : ikalibr/AWR1843RadarScan

```

clepz, 2025-06-03:
Thank you for your response. I'll continue to investigate the problem on my end. However, before I do, I have a few more questions:

1. In the River Dataset video, I noticed several figures attached to the walls in the office environment. Are these used to enhance reflectivity, or do you employ any specific materials/methods for that purpose?
2. I've been using TI's official ROS driver directly. Regarding this:
  2.a. Was any specific synchronization applied between the radars? The paper mentions "_software synchronized_," but I'm unclear on the exact meaning in this context. (I suspect if radars affect themselves. I tried to use hardware triggering with frameTriggerDelay config, however couldn't make it work.)
  2.b. Did you use the `1843_3d.cfg` sensor configuration from the ROS driver as is, or did you use a different configuration?
  2.c. I see the `AWR1843BOOST_CUSTOM` message type is present in `ikalibr`. I checked the repository you forked for this but couldn't find any additional commits beyond the message definition. Because of this, I used the official TI repository. Could this be causing a difference?
3. When I run the calibration repeatedly without changing anything, I get different results each time. What could be the reason for this variability?

Thanks again for your help.

Unsigned-Long, 2025-06-04:
The images provided in the River dataset are only meant to illustrate motion patterns. iKalibr is a targetless calibration method and does not require any artificial markers for assistance.

"Soft synchronization" refers to synchronization at the software level, which can be achieved using the official driver. In iKalibr, we define various message types (https://github.com/Unsigned-Long/iKalibr/tree/master/msg), which only serve to constrain the message format.

When the calibration result itself is not very accurate, the fact that repeated calibrations yield different results is not very meaningful, as the solution space is inherently uncertain.

I think you might want to check out the video from this repository (https://github.com/Unsigned-Long/RIs-Calib). It was the predecessor of IMU-radar calibration in iKalibr (although its functionality has now been fully integrated into iKalibr).