Unsigned-Long/iKalibr #34: lidar2rtk extrinsic标定

pedestrain123, 2025-07-03:
您好请问这个仓库适合固态速腾m1p激光雷达与组合惯导rtk之间的外参标定吗？期待您的回复

Huashuijingying, 2025-07-03:
我记得对速腾只能支持机械式激光雷达，详见https://github.com/Unsigned-Long/iKalibr/pull/28。另外，根据readme介绍，应该还不不支持标定rtk。

Huashuijingying, 2025-07-03:
我记得对速腾只能支持机械式激光雷达，详见https://github.com/Unsigned-Long/iKalibr/pull/28 。另外根据项目readme介绍，应该不支持对rtk标定

Huashuijingying, 2025-07-03:
对速腾雷达的支持只在机械式激光雷达helios-16上测试过，详见https://github.com/Unsigned-Long/iKalibr/pull/28 ，您可以在速腾的固态激光雷达上测试一下是否可用。另外根据项目readme介绍，应该不支持对rtk标定

pedestrain123, 2025-07-03:
> 对速腾雷达的支持只在机械式激光雷达helios-16上测试过，详见[#28](https://github.com/Unsigned-Long/iKalibr/pull/28) ，您可以在速腾的固态激光雷达上测试一下是否可用。另外根据项目readme介绍，应该不支持对rtk标定

后期会考虑对这个标定工具箱加入激光雷达与rtk的标定吗？感觉现有方法标定效果都很一般，而且开源资料相对较少

Unsigned-Long, 2025-09-22:
感谢您的宝贵建议！激光雷达与 RTK 的标定确实是不少实际场景中会遇到的需求。不过目前RTK 的定位精度在动态环境或信号复杂场景下稳定性可能不足，这会直接影响标定基准的可靠性。从实际效果来看，在中小尺度场景下，使用精密量具（如激光测距仪、全站仪等）进行人工标定，反而能获得更稳定的基准数据。RTK定位精度2-5cm，精度相比lidar等不高，手量杆臂是不是会更好。

closed 2025-09-22 by Unsigned-Long