Unsigned-Long/iKalibr #38: unknown error happened: 'yaml-cpp' bad conversion

ZeyaddMahmoudd, 2025-09-21:
Thank for this great work.
I am tryin to use iKalibr on my custom dataset with this config file which i find correct:
---
Configor:
  DataStream:
    IMUTopics:
      - key: "/imu"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/data/my_data_from_kalibr/imu-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
    CameraTopics:
      - key: "/camera/left/image_raw"
        value:
          Type: "SENSOR_IMAGE_GS"
          Intrinsics: "/data/my_data_from_kalibr/cam_l_intri.yaml"
          Weight: 50.0
          TrackLengthMin: 10
          ScaleSplineType: LIN_POS_SPLINE
      - key: "/camera/right/image_raw"
        value:
          Type: "SENSOR_IMAGE_GS"
          Intrinsics: "/data/my_data_from_kalibr/cam_r_intri.yaml"
          Weight: 50.0
          TrackLengthMin: 10
          ScaleSplineType: LIN_POS_SPLINE
    RadarTopics: []
    LiDARTopics: []
    RGBDTopics: []
    ReferIMU: "/imu"
    BagPath: "/data/my_data_from_kalibr/dynamic.bag"
    BeginTime: 5
    Duration: 23
    OutputPath: "/root/iKalibr_dev/src/ikalibr/data/my_data_from_kalibr"
  Prior:
    GravityNorm: 9.8
    SpatTempPrioriPath: ""
    OptTemporalParams: true
    TimeOffsetPadding: 0.05
    ReadoutTimePadding: 0.04
    MapDownSample: 0.1
    KnotTimeDist:
      SO3Spline: 0.05
      ScaleSpline: 0.05
    NDTLiDAROdometer:
      Resolution: 0.5
      KeyFrameDownSample: 0.1
    LiDARDataAssociate:
      PointToSurfelMax: 0.1
      PlanarityMin: 0.6
    CauchyLossForRadarFactor: 0.1
    CauchyLossForLiDARFactor: 0.5
    CauchyLossForCameraFactor: 1.0
  Preference:
    UseCudaInSolving: false
    Outputs:
      - LiDARMaps
      - VisualMaps
      - RadarMaps
      - VisualLiDARCovisibility
      - VisualKinematics
      - ColorizedLiDARMap
      - AlignedInertialMes
      - VisualReprojErrors
      - RadarDopplerErrors
    OutputDataFormat: YAML
    ThreadsToUse: -1
    SplineScaleInViewer: 3.0
    CoordSScaleInViewer: 0.3
but it always give me this error unknown error happened: 'yaml-cpp: error at line 29, column 11: bad conversion'
even if line 29 is commented or have another content.
can you help me how to solve it

Unsigned-Long, 2025-09-22:
The following configuration file works properly on my side, and the YAML parsing is correct. Perhaps you could try it once more on your end.
```yaml
Configor:
  DataStream:
    IMUTopics:
      - key: "/imu"
        value:
          Type: "SENSOR_IMU"
          Intrinsics: "/data/my_data_from_kalibr/imu-intri.yaml"
          AcceWeight: 17.68
          GyroWeight: 57.66
    CameraTopics:
      - key: "/camera/left/image_raw"
        value:
          Type: "SENSOR_IMAGE_GS"
          Intrinsics: "/data/my_data_from_kalibr/cam_l_intri.yaml"
          Weight: 50.0
          TrackLengthMin: 10
          ScaleSplineType: LIN_POS_SPLINE
      - key: "/camera/right/image_raw"
        value:
          Type: "SENSOR_IMAGE_GS"
          Intrinsics: "/data/my_data_from_kalibr/cam_r_intri.yaml"
          Weight: 50.0
          TrackLengthMin: 10
          ScaleSplineType: LIN_POS_SPLINE
    RadarTopics:
    LiDARTopics:
    RGBDTopics:
    ReferIMU: "/imu"
    BagPath: "/data/my_data_from_kalibr/dynamic.bag"
    BeginTime: 5
    Duration: 23
    OutputPath: "/root/iKalibr_dev/src/ikalibr/data/my_data_from_kalibr"
  Prior:
    GravityNorm: 9.8
    SpatTempPrioriPath: ""
    OptTemporalParams: true
    TimeOffsetPadding: 0.05
    ReadoutTimePadding: 0.04
    MapDownSample: 0.1
    KnotTimeDist:
      SO3Spline: 0.05
      ScaleSpline: 0.05
    NDTLiDAROdometer:
      Resolution: 0.5
      KeyFrameDownSample: 0.1
    LiDARDataAssociate:
      PointToSurfelMax: 0.1
      PlanarityMin: 0.6
    CauchyLossForRadarFactor: 0.1
    CauchyLossForLiDARFactor: 0.5
    CauchyLossForCameraFactor: 1.0
  Preference:
    UseCudaInSolving: false
    Outputs:
      - LiDARMaps
      - VisualMaps
      - RadarMaps
      - VisualLiDARCovisibility
      - VisualKinematics
      - ColorizedLiDARMap
      - AlignedInertialMes
      - VisualReprojErrors
      - RadarDopplerErrors
    OutputDataFormat: YAML
    ThreadsToUse: -1
    SplineScaleInViewer: 3.0
    CoordSScaleInViewer: 0.3
```

closed 2025-09-22 by Unsigned-Long

ZeyaddMahmoudd, 2025-09-22:
same error is there:
...
[C]-[358]-[10:31:35.442] unknown error happened: 'yaml-cpp: error at line 29, column 11: bad conversion'

Unsigned-Long, 2025-09-22:
Please ensure that your `iKalibr` is of the latest version (e.g., by pulling updates via `git pull`).

ZeyaddMahmoudd, 2025-09-22:
i have pulled the docker image again, made sure to pull the latest version, but still having the same error 

[image]

ZeyaddMahmoudd, 2025-09-22:
I think this because yaml-cpp version incompatibility, can you tell me which one do you use please?

Unsigned-Long, 2025-09-22:
> i have pulled the docker image again, made sure to pull the latest version, but still having the same error
> 
> <img alt="Image" width="1917" height="921" src="https://private-user-images.githubusercontent.com/134887267/492251500-c5453fcc-30de-425e-86f3-ba6aa15a7e4c.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg1NDE4MjgsIm5iZiI6MTc1ODU0MTUyOCwicGF0aCI6Ii8xMzQ4ODcyNjcvNDkyMjUxNTAwLWM1NDUzZmNjLTMwZGUtNDI1ZS04NmYzLWJhNmFhMTVhN2U0Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyMlQxMTQ1MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZjE4N2NhZjNmYjQxNDcyNDdhY2ViNmU1OWNjOGNlYjdlOGRjYjg1M2MyNzdmZjg4MjQzMTY2MTI4MTRkNmY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.DSOZ4pL7rFyMZSdIyXW-Qst--r8vwjbz3-6JF130Ouo">

It is evident that this YAML issue does not originate from loading the main configuration file, but rather arises when parsing the intrinsic parameter configuration files for the camera or IMU. Please revise your configuration files accordingly, referring to the example files provided at: https://github.com/Unsigned-Long/iKalibr/tree/master/config.

ZeyaddMahmoudd, 2025-09-22:
my cameras and imu yaml files follow these conventions.
this is cam_l_intri.yaml
`Intrinsics:
  # polymorphic pointers point to a base class and also point to objects of derived classes
  # don't care about this field
  polymorphic_id: 2147483649
  # the camera model type, supported models are:
  # (1) pinhole_brown_t2: k1, k2, k3, t1, t2
  # (2)  pinhole_fisheye: k1, k2, k3, k4
  polymorphic_name: pinhole_brown_t2
  ptr_wrapper:
    # don't care about this field
    id: 2147483649
    data:
      img_width: 1376
      img_height: 1100
      # fx, fy
      focal_length:
        - 909.668
        - 909.494
      # cx, cy
      principal_point:
        - 705.842
        - 570.178
      # k1, k2, k3, k4
      disto_param:
        - -0.171603
        - +0.120395
        - -0.02512
        - +0.000461439
        - +0.000150816`
 this is cam_r_intri.yaml
`Intrinsics:
  # polymorphic pointers point to a base class and also point to objects of derived classes
  # don't care about this field
  polymorphic_id: 2147483649
  # the camera model type, supported models are:
  # (1) pinhole_brown_t2: k1, k2, k3, t1, t2
  # (2)  pinhole_fisheye: k1, k2, k3, k4
  polymorphic_name: pinhole_brown_t2
  ptr_wrapper:
    # don't care about this field
    id: 2147483649
    data:
      img_width: 1376
      img_height: 1100
      # fx, fy
      focal_length:
        - 908.868
        - 909.06
      # cx, cy
      principal_point:
        - 698.437
        - 545.039
      # k1, k2, k3, k4
      disto_param:
        - -0.171725
        - +0.115331
        - -0.020063
        - +0.000269992
        - -0.000167839z`
and this is imu-intri.yaml which is yours as you stated it will get optimized in the visual inertial calibration
`# iKalibr: Unified Targetless Spatiotemporal Calibration Framework
# Copyright 2024, the School of Geodesy and Geomatics (SGG), Wuhan University, China
# https://github.com/Unsigned-Long/iKalibr.git
#
# Author: Shuolong Chen (shlchen@whu.edu.cn)
# GitHub: https://github.com/Unsigned-Long
#  ORCID: 0000-0002-5283-9057
#
# Purpose: See .h/.hpp file.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
# * Redistributions of source code must retain the above copyright notice,
#   this list of conditions and the following disclaimer.
# * Redistributions in binary form must reproduce the above copyright notice,
#   this list of conditions and the following disclaimer in the documentation
#   and/or other materials provided with the distribution.
# * The names of its contributors can not be
#   used to endorse or promote products derived from this software without
#   specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
# CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
# SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
# INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
# CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
# ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.

Intrinsics:
  # polymorphic pointers point to a base class and also point to objects of derived classes
  # don't care about this field
  ptr_wrapper:
    id: 2147483649
    data:
      ACCE:
        # 3-dime constant bias vector of the accelerator
        BIAS:
          r0c0: 0
          r1c0: 0
          r2c0: 0
        # mapping matrix, composed of scale factors and non-orthogonality factors
        # [      s_x,  gamma_xy, gamma_xz;
        #  -gamma_xy,       s_y, gamma_yz;
        #  -gamma_xz, -gamma_yz,      z_z]
        MAP_COEFF:
          r0c0: 1
          r1c0: 1
          r2c0: 1
          r3c0: 0
          r4c0: 0
          r5c0: 0
      GYRO:
        # 3-dime constant bias vector of the gyroscope
        BIAS:
          r0c0: 0
          r1c0: 0
          r2c0: 0
        # mapping matrix, composed of scale factors and non-orthogonality factors
        # [      s_x,  gamma_xy, gamma_xz;
        #  -gamma_xy,       s_y, gamma_yz;
        #  -gamma_xz, -gamma_yz,      z_z]
        MAP_COEFF:
          r0c0: 1
          r1c0: 1
          r2c0: 1
          r3c0: 0
          r4c0: 0
          r5c0: 0
      # rotation misalignment between the accelerator and the gyroscope (from frame {A} to frame {G})
      SO3_AtoG:
        qx: 0
        qy: 0
        qz: 0
        qw: 1`